Mr-MIBonk/M.I.B._More-Incredible-Bash #687: MIB patch download down :(

zfonseca-1, 2025-05-01:
Hi all! I guess it’s been a few days like this now, but patch MIB patch site is down.

Can anyone gently share this patch?
MHI2_ER_POG11_K3299_MU0807

Thanks in advance!

Maltman3000, 2025-05-02:
Looking for this as well

zfonseca-1, 2025-05-03:
> Looking for this as well

Go to Telegram link - I got it from there

zfonseca-1, 2025-05-05:
> Full collection
> 
> https://mega.nz/file/GQg03AaQ#13wJKlvsfbuuFHQUxtHuhqlMShz-chDfNSotGMRzhq4

Thanks a lot - this is great

yuvrxxj, 2025-05-18:
> Full collection
> 
> https://mega.nz/file/GQg03AaQ#13wJKlvsfbuuFHQUxtHuhqlMShz-chDfNSotGMRzhq4

This is incredible, thanks! Crossposting this on reddit

tomasmachatapepsk, 2025-10-16:
hello Maltman3000. 
again is mibsolution = off
have you pls send to me patches archive?

nikolica91, 2025-10-28:
I would kindly ask if you could send me the link to all the patches. I would be very grateful. Thank you in advance.