Mr-MIBonk/M.I.B._More-Incredible-Bash #688: G.E.M has a minimised box and really small text

h4n1793, 2025-05-03:
I have a issue I tried updating my firmware on my mmi system it all was successful but now when I go into the G.E.M menu the text is really small and I can’t see anything to change any help would be appreciated thank you

h4n1793, 2025-06-17:
I figured it out do you have odbeleven

giohouse, 2025-06-17:
I don’t have obdeleven

h4n1793, 2025-06-17:
Do you have anything to connect to the car to change the long coding

giohouse, 2025-06-17:
For example?
I have the screen like this. 
do you resolve?
I think that I have changed something in GEM but I do not rember what. 

[image]

h4n1793, 2025-06-17:
Yes something like obd

Babu408, 2025-07-15:
[image]