Mr-MIBonk/M.I.B._More-Incredible-Bash #757: No patch found for MHIG_EU_AU_K1555 MU0415

phantom83, 2025-08-14:
Good evening! I updated the firmware on my MIB1 high to version MHIG_EU_AU_K1555 with the intention of being able to update the maps from version 2016/2017 to version 2024. Unfortunately, when I accessed the green menu with M.I.B., in the patch folder I noticed that there is no patch for this version. What can I do? I've attached the .log file.

[MU0415-JTB-07424.01.14859001FE.log](https://github.com/user-attachments/files/21764228/MU0415-JTB-07424.01.14859001FE.log)

1tonyex, 2025-08-15:
from the log file

Patch folder /net/mmx/fs/sda0/patches/MHIG_EU_AU_K1555_MU0415_PATCH NOT found on SD card

did  u copy the patch MHIG_EU_AU_K1555_MU0415_PATCH to the patches folder in the sd card?

1tonyex, 2025-08-15:
Try to find it here, user:guest , passwd: guest, https://mibsolution.one/#/1/14

1tonyex, 2025-08-19:
[image]

phantom83, 2025-08-19:
If you try to download this file you realize that there are several patches inside it but not the one of my firmware.

azizdaboussi768-lab, 2025-08-28:
Men2_eu_skgpx_p160l

phantom83, 2025-08-30:
> Men2_eu_skgpx_p160l

sorry what does that mean?